sparkfun/9DOF_Razor_IMU #4: v23 Firmware Crashing on As-Delivered Razor

gdevenyi, 2015-02-10:
I just wired up my razor and turned it on, the menu shows up, and I can interact with it, the magnometer and the gyro work properly (print out measurements), however, if I try and trigger the Accelerometer output, the device crashes:
```
9DOF IMU Firmware v23 
==========================
[1]Accelerometer: ADXL345 
[2]Magnetometer: HMC5883 
[3]Gyroscope: ITG-3200 
[4]Raw Output
[5]Change Baud Rate: 57600bps
[Ctrl+z]Toggle Autorun
[?]Help
<PRESSED 1>

x=-22772, y= -28, ITG: GOOD
HMC: GOOD
ADXL: GOOD

9DOF IMU Firmware v23 
==========================
[1]Accelerometer: ADXL345 
[2]Magnetometer: HMC5883 
[3]Gyroscope: ITG-3200 
[4]Raw Output
[5]Change Baud Rate: 57600bps
[Ctrl+z]Toggle Autorun
[?]Help
```

This cycle is accompanied by the LEDs doing their "4 blinks and off" for "good"

ToniCorinne, 2015-02-13:
Hey there - sorry to hear about this problem. It sounds like this might be more hardware based, so please contact our tech support team at techsupport@sparkfun.com. They should be able to help you out with the hardware fix easier than we could do on GitHub (and get you a replacement if needed :) ).

gdevenyi, 2015-02-13:
In a email thread with them too. Thanks.

gdevenyi, 2019-02-18:
Defective device replaced.

closed 2019-02-18 by gdevenyi