sparkfun/9DOF_Razor_IMU #7: samd21_sam_ba.atsln file not existing in the repository

lunarpulse, 2017-01-09:
I have tried to compile and learn from the firmware. However, the repository does not have the solution file for Atmel studio. In addition, there are some errors while compiling without the solution file.

First: non existing member of the volatile structures of U

Error		'volatile struct <anonymous>' has no member named 'TRCPT' at 9dof_bootloader_sparkfun\board_driver_usb.c	176	52	Build

Error		'volatile struct <anonymous>' has no member named 'STALLRQ'	\9dof_bootloader_sparkfun\board_driver_usb.c	307	51	Build

Both of them are located in USB_DEVICE_EPINTFLAG_Type struct in usb.h. But the compiler could not find.

I guess these compilation issues may be caused by the absence of the samd21_sam_ba.atsln file or other absent files which contain the right configuration for the compilation.

Furthermore, could you guide me how to correctly compile this in Atmel studio?

lunarpulse, 2017-01-11:
I have investigated a little more and I found out that the USB_DEVICE_EPINTFLAG_Type union contains three fields of structs, bit, vec, and reg. 'TRCPT' and 'STALLRQ' both are member of vec, NOT bit.
the codes was intended to get the status bits stored as a union but the samba_usb.c file did not implement the use of the union struct correctly. I am not sure this was used by arduino zero directly. At least this does not work on atmel and linux arm gcc compiler.

After fixing all the errouneous member indication in the samba_usb.c file, the file compiles but still many errors to correct during the compilation of other files.

santaimpersonator, 2019-07-12:
I don't really have any information beyond what is listed here (I primarily do minor TS):
https://github.com/sparkfun/9DOF_Razor_IMU/tree/master/Firmware/Bootloader