sparkfun/9DOF_Razor_IMU #9: Roll, Pitch, Yaw values changing counterintuitive

theonteg, 2017-03-15:
I have two issues with the Roll, Pitch, Yaw values returned from the RAZOR IMU M0.

1. Yaw values are different every time the IMU turns on.
Depending on the heading of the IMU the time it turns on the yaw 0 value "snaps" on the closest cardinal point.

2. Roll and Pitch values interchange depending the Yaw value.
Turning the IMU around it's Y mag axis while Yaw is 0, changes the Roll value
Turning the IMU around it's Y mag axis while Yaw is 90, changes the Pitch value

Are those two issues normal? Because they are counterintuitive to me...

dahlsrud, 2017-03-16:
1. My experience is the same. I am not activating the  magnetic compass, so in that case the IMU does not have a fixed reference point for Yaw and just initializes it during startup to wherever it is pointing. The gyros can only detect relative movement, and the accelerometers can't detect Yaw. I guess the IMU *could* fix Yaw towards magnetic north at startup, but during testing I have never achieved that. For Pitch and Roll it is different, because the IMU can easily fix them to the horizontal plane by looking at the accelerometer and determining the direction of gravity.
If you want a fixed Yaw from every boot I guess you must add some extra code yourself by calculating the magnetic compass heading and compensating gyro Yaw against that. But I'd like to know more info about that myself.

2. In my understanding, this is a bug. Wikipedia has another formula for calculating Euler from Quaternions which works as expected:
https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles

theonteg, 2017-03-16:
1. Does enabled mag log means that it is enabled? 
I was hoping to avoid writing my own code for solving this (so that the solution is merged to the official firmware) but I will try.

2. Used this (Wikipedia's) approach yesterday but I have one problem. Pitch in some directions freezes at 90 degrees no matter the movement. I don't believe that there is something wrong with the code but I will check it thoroughly this weekend.

dahlsrud, 2017-03-16:
1. I never worked with it, but I suppose so yes.
I fear that we are lacking a good calibration code for the compass and that it involves having to rotate the IMU physically in 8-circles to gather calibration values for the earth magnetism in a given location.

Btw, the current original firmware also contains a bug which multiplies Pitch with 2 erroneously.

2. I posted working firmware code in this thread:
https://forum.sparkfun.com/viewtopic.php?f=14&t=45516
It contains its own routine to convert Quaternions to Euler angles. It is working great for me. 
You might have to modify the output format for your needs, though.