sparkfun/9DOF_Razor_IMU #13: Does the ROS driver razor_imu_9dof work for this model?

tony1098, 2017-12-07:
The razor_imu_9dof (http://wiki.ros.org/razor_imu_9dof) only mentions that it supports SEN-10736 and SEN-10125. Does it support this model as well?

lebarsfa, 2018-04-04:
Some support has been recently added for the M0 but using a modified firmware, see https://github.com/KristofRobot/razor_imu_9dof

improve100, 2018-04-10:
yes

santaimpersonator, 2019-03-14:
Unfortunately, I am not absolutely sure since the [SEN-10736](https://www.sparkfun.com/products/retired/10736) used completely different sensors than the MPU-9250 IMU. However, the resource [https://github.com/KristofRobot/razor_imu_9dof](https://github.com/KristofRobot/razor_imu_9dof) linked by [lebarsfa](https://github.com/lebarsfa) seems to indicate it is.

For further assistance, I will need to redirect you to the contact information on that wiki page:

> **Maintainer: Kristof Robot <krirobo AT gmail DOT com>**

who is probably better suited to answer further questions regarding their code.

closed 2019-03-14 by santaimpersonator