sparkfun/9DOF_Razor_IMU #26: IMU I2C address should be breakout

LucaCerina, 2019-10-07:
Hello,
I wanted to use the sparkfun Razor to record long-term accelerations. However, the default I2C address is 0x68 which conflicts with almost every decent I2C Real-Time Clock on the market.

I suggest that the default address should be changed to 0x69 or exposed with a small port on the breakout to be changed manually

improve100, 2019-10-07:
i have the board that you want.

LucaCerina, 2019-10-07:
which one?
Luca Cerina
----------------
Graduate Research Assistant @ NECSTLab
DEIB (Dipartimento di Elettronica, Informazione e Bioingegneria)
Politecnico di Milano

LucaCerina, 2019-10-07:
Of course I can rewire the circuit, but I would like to avoid the process on multiple samples just to have something that could be already available with an exposed soldering pad or something like that

improve100, 2019-10-07:
i think you can jump wire.or buy the board that you want

nseidle, 2019-10-07:
OP - good point. We should have broken out the I2C address pin to a jumper or GPIO so that the user could change the IMU address. Unfortunately it's hard wired to 0x68. We'll try to fix this on a future rev.

As an aside, the SAMD21 is clocked with a 32.768kHz xtal. You may be able to setup an RTC function directly on the micro without the need for an external RTC.

LucaCerina, 2019-10-08:
Fair enough, I will try also with that and see if the drift is ok.

closed 2019-10-08 by LucaCerina