sparkfun/9DOF_Razor_IMU #27: Not reading serial data if booting with SBC

msadowski, 2019-10-21:
Hi!

I have a weird issue that I've seen in two separate firmwares for this IMU board. What I'm observing is that if I connect the IMU to my USB port while the SBC is running everything seems to work well - I can read data, and with the default .ino sending a space character stops the stream (1).

If however I have the IMU plugged in to the SBC when I power up the SBC I read the IMU data just fine but I can't send anything to the board (2).

When debugging I was printing the serial available bytes (LOG_PORT.available()) in case 1 as I was sending characters I could see that available() bytes on the log port increased. In case of 2nd configuration the LOG_PORT.available() always returned 0.

Did anyone run into this? Any tips on how to approach this problem?

msadowski, 2019-10-22:
As a follow up test I've tested this with a trivial sketch that reads serial data and sends it back. I can confirm that this issue is also visible in this case:

```
int incomingByte = 0;    // for incoming serial data

void setup() {
    SerialUSB.begin(9600);    // opens serial port, sets data rate to 9600 bps
}

void loop() {
  // send data only when you receive data:
  if (SerialUSB.available() > 0) {
  
    // read the incoming byte:
    incomingByte = SerialUSB.read();
  
    // say what you got:
    SerialUSB.print((char)incomingByte);
  }
  
}
```

santaimpersonator, 2020-10-08:
What does SBC represent?

I've used the Razor on multiple computers and I've never had an issue retrieving data or controlling it from a serial terminal (emulator) after the computer boots. Have you tried debugging whatever the SBC is?

msadowski, 2020-10-09:
SBC is "Single Board Computer" in this context. Basically any time I was powering the machine with the board already plugged in I was having issues.

santaimpersonator, 2020-10-09:
I have never had any issues with the Razor working on a Raspberry Pi from boot up. However, I have never really been trying to control it through the serial (USB) connection. I was mostly testing it out as a VR headset concept: 
https://www.sparkfun.com/news/2882

Maybe it is a kernel/application specific issue communicating with the USB device? I have found screen to work consistently, don't forget to check your baud rate... I've seen some emulators display properly at double/quadruple the baud rate.
Since you are using an SBC... have you tried just connecting it to the serial pins directly?